leaflabs/libmaple #106: Installing on OSX 10.10.1 Yosemite

moskrc, 2015-01-24:
Hi,

I have a problem with 'make',

```
vit@myhost ~/projects/electronics/libmaple $ make
make: arm-none-eabi-gcc: Command not found
make: arm-none-eabi-gcc: Command not found
================================================================================

  Build info:
     BOARD:           maple_mini
     MCU:             STM32F103CB
     MEMORY_TARGET:   flash

  See 'make help' for all possible targets

================================================================================

  [CC]        ./libmaple/adc.c
arm-none-eabi-gcc: error trying to exec 'cc1': execvp: No such file or directory
make: *** [build/./libmaple/adc.o] Error 1
vit@myhost ~/projects/electronics/libmaple $
```

```
vit@myhost ~/projects/electronics/libmaple $ arm-none-eabi-gcc
arm-none-eabi-gcc: no input files
vit@myhost ~/projects/electronics/libmaple $ cc1
-bash: cc1: command not found
vit@myhost ~/projects/electronics/libmaple $ cc
clang: error: no input files
vit@myhost ~/projects/electronics/libmaple $
```

Thanks

bnewbold, 2015-01-29:
Hi @moskrc, it looks like your problem is what the output says: the arm-none-eabi-gcc toolchain can't be found on your $PATH.

The libmaple documentation (http://static.leaflabs.com/pub/leaflabs/maple-docs/latest/unix-toolchain.html) is very out of date on this topic, but you'll probably want to download the pre-compiled OS X toolchain (https://launchpad.net/gcc-arm-embedded), add it to your path (so you can run, eg, arm-none-eabi-strip from the command line and get help text and not a "not found" error), then try again.

bnewbold, 2015-02-26:
@mbolivar, please close this issue.

closed 2015-03-06 by mbolivar