leaflabs/libmaple #107: USART hangs on read overrun

ctz, 2015-03-06:
I have a program which alternately communicates on uart1 and bit-bangs GPIO to talk a certain protocol. When doing the bit-banging, interrupts are turned off.

This hangs very occasionally (~once an hour) on maple mini.

I attached a debugger and caught the hang in action: it's an interrupt flood to USART1 because `usart_irq` doesn't handle all the exceptions it enables.

The datasheet says:

> Bit 3 ORE: Overrun error
> This bit is set by hardware when the word currently being received in the shift register is
> ready to be transferred into the RDR register while RXNE=1. *An interrupt is generated if
> RXNEIE=1 in the USART_CR1 register. It is cleared by a software sequence (an read to the
> USART_SR register followed by a read to the USART_DR register).*

So while the current code will read DR and therefore clear RXNE, it never reads SR to clear ORE.  If there is an overrun it will never clear that exception and it will be continually reasserted forever. This is the cause of the hang I saw.

ctz, 2015-03-06:
https://gist.github.com/ctz/86b63fa8d95b42d04245

Patch which Works For Me™

mbolivar, 2015-03-08:
Thanks for this! Mind submitting a PR?

madias123, 2015-05-16:
ctz: How about join the new STM32-arduino community: http://stm32duino.com/ ?

bnewbold, 2015-06-14:
@mbolivar patch was merged, so this ticket should be closed now

mbolivar, 2015-06-15:
done, thanks

closed 2015-06-15 by mbolivar