leaflabs/libmaple #109: Freeze when erasing page and receiving serial data

Gregwar, 2015-03-10:
Hello,

I'm working on an application where I need to change the flash. So, I erase pages and write bytes on-the-fly. 

This is the erase page part:

```
    FLASH_BASE->CR |= FLASH_CR_PER;
    while (FLASH_BASE->SR & FLASH_SR_BSY);

    FLASH_BASE->AR = pageAddr;
    FLASH_BASE->CR |= FLASH_CR_STRT;
    while (FLASH_BASE->SR & FLASH_SR_BSY);
    FLASH_BASE->CR &= ~FLASH_CR_PER;
    
    return true;
```

When there is serial data incoming in the same time, this results in a complete freeze of the chip just after starting the erasing.

Any idea why?

Gregwar, 2015-03-11:
Looks like the uC doesn't freeze if I disable uarts before doing my operations and enable it again just after
I understand that a flash operation would stall the uC for like 20ms anyway, making me loose bytes on the uart. But I don't understand why it hangs forever.

mbolivar, 2015-03-11:
I have no idea. Perhaps bus contention oddness?

What happens if you put nop or WFI in the loop instead of a hard spin? (Unless you've turned it off, SysTick will wake you up if you go WFI).

Gregwar, 2015-03-11:
I've tried a lot of things, and it looks like there is a relation with the usart handling
One of the possibilities is that thr usart interrupt occur at some point for another reason than rx data
Looks like checking the RXNE bit in the SR usart registers before appending data to the rb prevents hanging

mbolivar, 2015-03-11:
e1314f6466102 just got merged, which might fix that. Can you try with today's master?

Gregwar, 2015-03-11:
Actually, this is what I did
I've also moved the uart interruption to the ram section

mbolivar, 2015-03-11:
OK, so is today's master working for you? I'll close if so.

Gregwar, 2015-03-11:
Yes, I confirm it's ok

closed 2015-03-11 by Gregwar