leaflabs/libmaple #111: Reducing heap size?

Gregwar, 2015-08-12:
Here:
https://github.com/leaflabs/libmaple/blob/master/libraries/FreeRTOS/utility/FreeRTOSConfig.h#L76

We can see that 8K RAM are allocated for the heap (this is like 40% of the ram). Can this be reduced safely? How much of this  heap are actually used?

mbolivar, 2015-08-13:
I'm tempted to just rip that code out, honestly. Are you using it?

Gregwar, 2015-08-13:
Absolutely not, I just saw that thing in the sizes
BTW, why is FreeRTOS here?

mbolivar, 2015-08-13:
It was submitted a while back and I took it. It's gone now.

closed 2015-08-13 by mbolivar

Gregwar, 2015-08-13:
You're changing something on master?

mbolivar, 2015-08-13:
Yes, sorry, pushed to the wrong remote :)

Gregwar, 2015-08-13:
Ok, thanks
Indeed, this really reduces the RAM usage (I was short in some projects)
Thanks again for all this work and the Maple mini which is a great tool
Do you plan to release a new version later with a bigger MCU like M4 serie?

mbolivar, 2015-08-13:
We've EOLed the line, actually, but I'm still taking patches as a spare time thing:

http://www.leaflabs.com/news/2015/3/27/maple-end-of-life-notice

mbolivar, 2015-08-13:
And thanks for the kind words; they're appreciated :)

Gregwar, 2015-08-13:
Did you know that the Maple Mini is now produced by many sellers, and available to prices as low as 
4$?
http://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20150813124323&SearchText=maple+mini

There is actually not so much alternatives for such product

Gregwar, 2015-08-13:
(I've tried the 4$ versions, there is some cost-killing on some components but it's still good)

mbolivar, 2015-08-28:
Hm, we did not know that. I'm glad the hardware lives on, but it's not nice of them to be using our company and brand name like it's an authentic product. Thanks for the heads up; we're looking into it.