leaflabs/libmaple #113: Why skip the first 3K of ram?

Gregwar, 2015-09-08:
Here:
https://github.com/leaflabs/libmaple/blob/master/support/ld/stm32/mem/sram_20k_flash_128k/mem-flash.inc#L3

Why the first 3K of ram are skipped?
I mean, why don't we start at 0x20000000 with 20K?

bnewbold, 2015-09-08:
I can't really remember, but maybe some version of maple-bootloader remained partially in RAM? http://static.leaflabs.com/pub/leaflabs/maple-docs/latest/bootloader.html#maple-rev-1

If you want to use all the memory and don't depend on the bootloader, you can use the JTAG target (even if you're using some other mechanism to upload, like a hardware ROM bootloader): https://github.com/leaflabs/libmaple/blob/master/support/ld/stm32/mem/sram_20k_flash_128k/mem-jtag.inc

rogerclarkmelbourne, 2016-09-12:
See https://github.com/rogerclarkmelbourne/STM32duino-bootloader

It does not use the first 3k of RAM

I think the reason was something to do with running the sketch from RAM but as on the STM32F193C8 the RAM is hardly big enough to hold even a blink sketch, that option has been removed from my version, and it free's up the 3k

Gregwar, 2016-09-14:
Yes I removed this skip too in my fork