leaflabs/libmaple #117: PWM not working on pin 9?

Gregwar, 2016-03-21:
It appear that pin 9 PWM doesn't work, the following code:

```c++
    HardwareTimer timer(2);
    timer.pause();
    timer.setCount(0);
    timer.setOverflow(1000);
    timer.setPrescaleFactor(1);
    timer.setMode(TIMER_CH1, TIMER_PWM);
    timer.setMode(TIMER_CH2, TIMER_PWM);
    timer.setMode(TIMER_CH3, TIMER_PWM);
    timer.setMode(TIMER_CH4, TIMER_PWM);
    timer.refresh();
    timer.resume();

    pinMode(9, PWM);
    pwmWrite(9, 500);
```

Doesn't produce a PWM output on pin 9
However, writing it using OUTPUT and digitalWrite works
Also, producing a PWM on pin 8, which is also on timer 2 works

Any idea? (I've tried on two different boards just in case)

mbolivar, 2016-03-23:
No idea, sorry. We'd gratefully take a patch, but since Maple is EOL won't be working on issues.

Gregwar, 2016-03-23:
OK
Do you have the setup to confirm or disconfirm the issue?
No problem if you can't, I will investigate
Thank you

Gregwar, 2016-03-29:
It actually appear that this is related to my bootloader (still not sure why)

closed 2016-03-29 by Gregwar